goebish/nrf24_multipro #3: CX10 Red (original) support

samuelpowell, 2015-10-06:
@goebish, would you like me to port [my old code](https://github.com/samuelpowell/cx10_redtx) to support the original CX-10?

goebish, 2015-10-06:
Maybe I'll add it (along with YD717 and variations) if someone ever needs it, but looks like this one has been discontinued for a long time. There is newer red pcb (using xn297) that has not been reverse engineered yet, I'm waiting for some stock TX spi dumps :)

goebish, 2015-10-06:
... but if you want to add it to this project, just wait a little, someone has ported my code to a cleaner library version ( https://github.com/Links2004/nrf24_multipro ), I'll merge it here once I've checked everything works correctly.

samuelpowell, 2015-10-06:
Understood, thanks!

closed 2015-10-06 by samuelpowell