goebish/nrf24_multipro #6: H8_3D ligth

pascallanger, 2016-01-20:
Hi Goebish,
This is not an issue and I don't know how to really raise that but here it is.
For H8_3D protocol and at least JJRC H22 there is a flag to control the quad ligths:
H8_3D_FLAG_LIGTH = 0x08, // Light on H22
if(Servo_AUX2)  packet[17] |= H8_3D_FLAG_LIGTH; //H22 light
Since the H8 3D seems really similar I think it should work as well.
- Pascal

goebish, 2016-01-20:
Hi, I think lights can't be switched off on the H8_3D, but I'll add this flag in MultiPro and DeviationTX, thanks :)

pascallanger, 2016-01-20:
Thanks for your hard work on protocols. It's only a small contribution...
The JJRC H22 TX does not have a button for ligth but playing with the flags I've discovered the functionnality.
By the way, I'm working on a similar project like this one which you may know. It reuses a lot of the protocols you've been working on and available on deviation: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module. There migth be some interrest to work together...
- Pascal

closed 2016-01-20 by pascallanger

goebish, 2016-01-20:
OK, LED flag added to H8_3D protocol on AUX1 (channel 5)
https://github.com/goebish/nrf24_multipro/commit/99f026161980b5fc859504b953f63732f47af27a