goebish/nrf24_multipro #9: Protocol

Chen-Leon, 2016-04-03:
HI
Can addin Q242 protocol ?

goebish, 2016-04-03:
Yes, that's a CX10 sub protocol, I'll add it to my todo list ;)

Chen-Leon, 2016-04-03:
Goodjob Thank's

H3adsho0ot, 2017-01-06:
@goebish  Have you tried to control a JJRC H20H instead of a normal H20? Im really interessted in the att hold function.

goebish, 2017-01-06:
The JJRC H20H is using yet another protocol, I need more captures to thoroughly understand it:
http://www.deviationtx.com/forum/protocol-development/4724-jjrc-h20-hex-quad?start=180#56649

H3adsho0ot, 2017-01-06:
Uh okay. If i get my i'll try to capture it but I think this might take some more weeks.

goebish, 2017-01-06:
No hurry, the H20H isn't on the top of my list yet, I want to add the CX10D and CX10WD first.

ExiterOne, 2017-02-09:
Help needed on adapter use on AT9S. Have Goebish nrf24 Multiprotocol adapter. Tried OpenTX, no support for AT9S. Not sure how i hooked up with trained port is correct. Used ground and out port on transmitter for what I hope is the PPM out. Going to try and look at signal when I find my O scope. Have heard it can be done.