goebish/nrf24_multipro #14: Help Binding...

befreestudios, 2016-10-03:
I've been trying to bind the E010 with the given instructions with no luck. Is there some kind of trick I'm missing or something? 

I've got the left stick pointed right, and the right stick pointing right and down. Double checked and the model settings are correct too.

Any help is greatly appreciated... thx

goebish, 2016-10-03:
Make sure your transmitter is in TAER mode.
There was an issue with the first commits of the E010 protocol, some TX IDs didn't work, this is fixed now, can you update your module with the latest code?

If you can't update the firmware, just try to start the transmitter with rudder stick pointing left, that will assign a new ID to the module, then retry to bind to the E010 (disconnect / reconnect battery on the E010 between each tries, that's important).

Email me if you need more instructions ;)

befreestudios, 2016-10-03:
Got it! Didn't realize I needed to hold the sticks in the right position _**while**_ turning transmitter on. Also, I must have the latest firmware 'cause it rebinds everytime now. Woo Hoo!

goebish, 2016-10-03:
No problem, glad that you got it to work, I'm closing this issue ;)

closed 2016-10-03 by goebish

cloudes79, 2016-11-26:
hi I found the same problem. I still can't bind with E010. i try to bind with H8 it's ok. but with E010 can't.

I also try to change ID. left stick go left down. not turn off the taranis and then disconnect/reconnect E010 and right stick taranis go right down...but still can;t connect

please help

goebish, 2016-11-26:
This has nothing to do with the original issue, but which nrf24l01 module are you using ?
It's been found that xn297 emulation at 250kbps doesn't work with some (fake ?) nrf24l01 chips, hence it works with the H8 (1 Mbps) but not the E010 (250 kbps).

cloudes79, 2016-11-26:
[image]

goebish, 2016-11-26:
I've built 100+ modules with this nrf24l01 module with 100% success with the E010 ...
Looks like this one was made by me, if it's been sent after the month of August then it has already been tested with an E010.
Can you try with this .eepe for OpenTX (no ntick combination required at startup) :  https://www.dropbox.com/s/5m09lgegttuh7ao/e010.eepe?dl=1

cloudes79, 2016-11-26:
i once bind with my old taranis. but it was broken. i buy the new taranis now i'm using it...can't bind...is it has to do with the taranis firmware also?

goebish, 2016-11-26:
No, it should work with any version of OpenTX, have you tried with the .eepe I linked ?

cloudes79, 2016-11-26:
already. but still cant not bind

goebish, 2016-11-26:
When did you receive this module ? Do you have a 3.3 Volt FTDI so you can update its firmware ?
Though if it has already worked with your E010 there's no reason it wouldn't work now ...

cloudes79, 2016-11-26:
my E010 also new

goebish, 2016-11-26:
I'm not sure what the problem is, but if you ordered your module before the 18th of October I'd try to update the module's firmware as I made a few changes to the E010 protocol on this date.

cloudes79, 2016-12-04:
hi do you have tutorial to update the module?

goebish, 2016-12-04:
To update the module you will need a 3.3 Volt FTDI dongle:
http://bit.ly/FTDI_dongle

Connect it to the module, make sure the jumper is set to 3.3 Volt position:
http://imgur.com/omcbZmJ
(use a rubber band to have a good contact ...)

Download and extract the latest firmware source code:
https://github.com/goebish/nrf24_multipro/archive/master.zip

Get the latest Arduino IDE version:
https://www.arduino.cc/en/Main/Software

In the Arduino IDE, load nrf24_multipro.ino
In the tools menu, select Arduino pro mini with ATmega328p 5V/16MHz processor and the COM port the dongle is mapped to
In the toolbar, click "Upload"
Et voila, it's updated.

cloudes79, 2016-12-04:
i have this ftdi. can i use this? how to connect the port?
[image]

goebish, 2016-12-04:
No, looks like this one is 5 Volt, that would kill the nrf24.

techcoast, 2017-02-11:
I'm having the same issue. I guess I'll have to find that module and try an update.