goebish/nrf24_multipro #17: Eachine E010 Multiple Transmitter Interference

Tschingis, 2016-10-18:
Hi,

We have the problem that if we fly two eachine E010 at the same time with two multipro enabled taranis radios, that we get random failsafes on both copters. It doesn't matter if the copter is far away from its assigned transmitter or not. We also tested one multipro enabled taranis radio and one stock radio which is included by the eachine E010 which showed no such error.

Any idea how to fix this ?

Greets,
    Markus

goebish, 2016-10-18:
First, try to change TX ID on one of the modules (start with rudder stick to the left).
This protocol is not fully cracked, I found how to use a semi arbitrary ID but for now they're all using the same set of RF frequencies. I'm still waiting for a few more TX ID / Frequencies sets to try to fully understand how it works: http://www.deviationtx.com/forum/protocol-development/6307-eachine-e010-inductrix-clone-data-captures?start=160#54330

Tschingis, 2016-10-18:
Hi, thanks. I will give it a try and play around with two different sets of fly channel sequences to get our radios working.

goebish, 2016-10-18:
Some change to the code will be required for that as I only implemented a special case where bind and data frequency sets are the same.
Anyway, I've got 3x E010 now, I'll make some tests as well, worst case I'll just implement a list of known ID / Channels which you will be able to select in the code.

Tschingis, 2016-10-18:
If i correctly understand the code in MJX.ino, i can simply use another (hardcoded) rfchannel map and txid in mjx_send_packet when bind is not set to one. So far I did not understand, why mjx_txid is three bytes in code and two bytes in the above forum post.

goebish, 2016-10-18:
Yes I think you understand correctly. mjx_txid is 3 byte in code because that's the size of the MJX X600 ID, but E010 is 2 bytes only (well, the 3rf byte is always 00 ...).

goebish, 2016-10-18:
Also, if you've a logic analyzer, your help at finding the proper txid / rf channel algo by providing captures of the spi bus from some stock TXs is welcome :)

goebish, 2016-10-18:
I've just added 2 extra id / freq sets in the post above, I still can't see a clear pattern, if any. I'm starting to wonder that there may be only 16 sets of hardcoded frequencies, but more captures from stock TXs are required to confirm that. Meanwhile, I'll just implement the 8 known id / freqs I have, that should fix your issue.

goebish, 2016-10-18:
I just pushed a temporary fix, there are 8 txid / channels sets hardcoded now.
It will choose one "randomly" depending on your transmitterID[0].

Tschingis, 2016-10-18:
Thats cool, i just wanted to start coding a similar approach.  I looked at the txid -> channel map alot at work *cough* and tried different transform algos like and,or,xor,add,sub for the txids to see some kind of pattern that appears in the channel map. The pattern of the channel map (in nibbles) seams to be:
([A][X]) ([B][Y]) ([A+1][X]) ([B+1][Y])
()  : Byte , [ ] : Nibble 
But how to select A,B,X and Y nibbles depending from the txid. I also thought that there is a upper and lower limit for a channel like  2E and 4E so that frequency is tight settled around a crappy antenna often used in such devices. 
I have seen no relation so far and since your newest extensions shows that  number 6 and 7 have equal channel maps it could be really just a hardcoded table.

Tschingis, 2016-10-18:
I've tested the new code with my single E010 and it worked so far (i think i tell nothing new). I will test the next days in a tiny whoop session with multiple E010 at once if the problem is fixed.

goebish, 2016-10-18:
Note that there's still a risk of collision as there are only 7 sets of frequencies, just assign a new arbitrary transmitter ID to one module (start with rudder to the left) if interferences still occur.

Tschingis, 2016-10-21:
Your fix works perfectly. Yesterday we tested around 15 batteries on each eachine e010 in parallel and we had no single failsafe.

closed 2016-10-21 by Tschingis

goebish, 2016-10-21:
Thanks for reporting, I'll also port the fix to DeviationTX.