goebish/nrf24_multipro #18: High rates Eachine E010

Rikbruggink, 2016-12-07:
Hi Goebish,

When i bind a E010 to a module based on this code it feels like the rates are set to low with no option change it to high. I dont find the references in the source code to set the high rate on a eachine e010 machine.

goebish, 2016-12-07:
Rate is fixed to high:
https://github.com/goebish/nrf24_multipro/blob/master/nRF24_multipro/MJX.ino#L137

Maybe you need to setup your endpoints ?

Rikbruggink, 2016-12-07:
hmm, let me check that. It doesn't feel like high at the moment. I did see that packet being sent didn't recognize it as the 'high' rate. I will further investigate. Closing this issue until it results in a 'bug'. Is there a rcgroups thread where these questions are more appropriate? Github issues is not the best place for general usage issues of the source code.

closed 2016-12-07 by Rikbruggink

goebish, 2016-12-07:
Here's the "unofficial" support thread:
https://www.rcgroups.com/forums/showthread.php?2367400-CX-10-JR-Module-(Green-Blue-PCBs)
Please tell which transmitter you're using.
I test all my modules with an E010 before sending them, so I'm sure there's no issues with this quad ;)

Rikbruggink, 2016-12-07:
The transmitter is a Taranis X9D with a multi-tx module i got as a holiday gift from BG. From the pdf i noticed that it was using a modified version of this firmware. AETR vs TEAR. I don't have the source code so i cant verify that the packet 14 bit is set to the referenced file. I believe it is. It feels a bit sluggish at the moment. The RCgroups thread i found also looking for answers, didn't think it was also intended for general usage questions of the nrf24 firmware.

goebish, 2016-12-07:
Sorry, but I can't give support for the BG module, I'm not affiliated to this module in any way.
And if it's using my firmware without giving the source code then it's in violation of the GPL.

kilrah, 2016-12-07:
If you're talking of the "usual" BG multiprotocol module you're at the wrong place.
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module

goebish, 2016-12-07:
He mentioned a pdf that's a modified copy of my readme.md, I think he has the nrf24 only module, knockoff of the nrf24-multipro.

silver13, 2016-12-07:
I don't think there is a flashing port on the bg module.

It is also disappointing they chose to ignore licensing. Perhaps someone should notify the FSF.

Rikbruggink, 2016-12-08:
﻿I understand that you can't support this module. I doubt they wil release the firmware but who knows. Wil try to flash this firmware to see if it works. It has a mega328 so it should compile.

tbaumann, 2016-12-08:
I think he means BangGood with BG.
http://www.banggood.com/Multiprotocol-TX-Module-For-Frsky-X9D-X9D-Plus-X12S-Flysky-TH9X-Transmitter-p-1097688.html

Looks like a derivative of @goebish module. But who knows.
Supportability? Who knows.