goebish/nrf24_multipro #19: Cheerson CX-10W?

koalapenguins, 2016-12-23:
Hi! Awesome repository you're sharing with the world :D

Has anyone tried this library with the Wifi version of the Cheerson CX-10, particularly the CX-10W?

Or, how well would this library work with other CX-10 variants?

Cheers

goebish, 2016-12-23:
I'm not sure for the CX10W, but the CX10WD & CX10D are using different protocols, they're not compatible with nrf24-multipro yet, maybe in a few weeks as I will implement them into DeviationTX soon.
https://www.deviationtx.com/forum/protocol-development/6448-cx-10d-traces

goebish, 2016-12-23:
According to the forums, the CX10W is using the same protocol than the CX10A (CX10 "Blue PCB") so it should be compatible with nrf24-multipro already.

koalapenguins, 2016-12-23:
Oh Nice! Will look into your projects more. Can't wait to start tinkering!

Thanks! :D

closed 2016-12-23 by koalapenguins