goebish/nrf24_multipro #22: FuriBee H36

timmyhadwen, 2017-01-19:
Seems to be a different protocol to the JJRC H36 :(

Ill try and get some caps :)

goebish, 2017-01-19:
Looks like there's nothing much to capture ... Seems the MCU and the RF are in the same chip, no external SPI bus ...

silver13, 2017-01-19:
have you been able to see any marking on the cpu? Maybe there will be a datasheet for this one ( unrelated to the protocol issue )

goebish, 2017-01-20:
Seems there's no marking except on the MPU:
https://www.rcgroups.com/forums/showpost.php?p=36437721

silver13, 2017-01-20:
Cool, that's the only pic I have found so far, not really very likely to find the datasheet for it, I hope the protocol gets reversed, it would be a challenge if both tx and rx are single ic

goebish, 2017-01-20:
Edit: the Tx might have a separate RF chip on the other side of the board, looks like there's a 5 pin header.

silver13, 2017-01-20:
does the tx chip have any markings? Anyway I have one in mail, I guess I'll check when It arrives