goebish/nrf24_multipro #23: E010 new protocol?

spaivaras, 2017-01-23:
Thx for the great work!
But can some one ensure that this is working with E010?

I have a http://www.banggood.com/Eachine-E010-Mini-2_4G-4CH-6-Axis-Headless-Mode-RC-Quadcopter-RTF-p-1066972.html?rmmds=myorder

And http://www.banggood.com/Multiprotocol-TX-Module-For-Frsky-X9D-X9D-Plus-X12S-Flysky-TH9X-Transmitter-p-1097688.html?rmmds=myorder

I tried with stock FW (which seems exact like this by pinout even the led pin and flashes are the same, sadly was not able to compare as atmega had lock bits set). Soldered out NRF24L01 which comes with the module, used 2 additional NRF24L01. Compiled this fw with every possible output power settings. But was not able to bind to E010 with any modifications (nor the original board).

Transmitter is FrSky Taranis X9D plus, PPM and channels seems to be ok, as i get led blinks while selecting protocol.

I even sniffed original remote and this module with SDR (HackRF), frequencies are quite close (not ideally same) sadly don't image how to demodulate to check the packet data.

So TL;DL did anybody used this code to successfully bind to E010? thanks

goebish, 2017-01-23:
My modules are working perfectly with the E010, no idea for the banggood 
knockoff ...

/ Goebish

spaivaras, 2017-01-23:
Well eachine e010 itself is a knockoff :) In any case, thx for the confirmation. Sadly the toy controller chips are "sanded", but maybe some pocking with a scope could reveal something. Thx again, this probably can be closed (didn't expect such fast response :D)

goebish, 2017-01-23:
That should be a xn297l
https://www.deviationtx.com/forum/protocol-development/6307-eachine-e010-inductrix-clone-data-captures

spaivaras, 2017-01-23:
Thank you for the pointers. And by knockoff you probably ment the tx boards, not the quads :D (silly me) in any case after i removed the nrf, it only was left with an atmega328p, a 16Mhz clock, and a 3.3V reg. So in any case, the code should do the same :)

commit a2c9cda3 (2016-10-18): [E010] Replace semi random txid with a known working set of txid / rf channels, should fix #23
closed 2017-01-23 by goebish