goebish/nrf24_multipro #24: Rarely binding with CX-10 Blue PCB

heidtn, 2017-02-06:
I've managed to get my CX-10 to bind via nrf24l01 only at one point.  It connected for a few times after the initial binding then I put it away.  Took it out the next day and was no longer able to get it to bind.  

I'm receiving data back from the quad that looks like this (in hex):
`AA, E3, 25, B1, AF, 20, D9, 32, 00, 00, 00, DC, 05, E8, 03, DC, 05, 11, 00, 00`

As you can see the bind item is set to 0 no matter what I do.  I receive this packet exactly once until I reset the arduino.  Resetting the quad does not send further data.  Any thoughts here?

The only change I've made is using the hardware SPI instead of the soft SPI, could that cause issues?

silver13, 2017-02-06:
check the CE pin of the nrf24 is set correctly just in case

heidtn, 2017-02-06:
Triple checked.  Same output.  Thanks for the quick response!

silver13, 2017-02-06:
Is there any chance the nrf24 module was powered with over 3.6V? The rx part usually breaks first, and I had one which did exactly this, although others have broken in other ways too.

heidtn, 2017-02-06:
Just checked with a multimeter, power is fine, but the pins (MOSI, MISO) are powered at 5V.  Could that break it?  It's weird that I'm getting a response from the quad though right?

silver13, 2017-02-06:
The non power pins are 5V tolerant, it would need 5V on the power pin to break.

Daniel-1276, 2017-10-14:
Same issue here with commercialy available tx module from Banggood. Multiple power cycles of my Taranis enables rebinding. No idea what's causing this.