goebish/nrf24_multipro #25: eachaine h8 mini (blue board)

alexkwinto, 2017-02-20:
Hi, 

is seems that h8 mini (blue board) has changed the protocol. Is there any way to check it?

Alex

goebish, 2017-02-20:
Where did you read that ?
All the blue boards I've seen are using the same protocol than the green ones.

alexkwinto, 2017-02-20:
just an assuption by failing to connect =) that is why i'm looking for a way to check it.

goebish, 2017-02-20:
Have you checked your module with another machine ?

alexkwinto, 2017-02-20:
no, i have no other copters to check. so yes the problem might be in my build

goebish, 2017-02-20:
Yes, triple check your wiring, make sure you give only 3.3 Vcc to the nrf24l01, and depending of the transmitter you're using you probably have to adapt the level of the PPM output to 0-3.3V.

alexkwinto, 2017-02-20:
i'm overrided protocol selection to bind to copter regardless the ppm values. is there a way to check arduino-nrf24 connection?

goebish, 2017-02-20:
yes, check if NRF24L01_Reset() returns 1.

alexkwinto, 2017-02-20:
TY, i will check it in few days and will let you know.

alexkwinto, 2017-02-20:
hi again, NRF24L01_Reset() returns zero. im using arduino uno to debug and printing to port monitor

alexkwinto, 2017-02-20:
ok i found the issue! THank you kindly!

goebish, 2017-02-20:
Cool, can you explain the issue you had ?

alexkwinto, 2017-02-20:
connection order( but now i can see an other issue - on low throlette the motors spinnig with interrupts

alexkwinto, 2017-02-20:
also on hight trollette, can this be a connection interrupt or something?

goebish, 2017-02-20:
Check your antenna, also, if you're using a nrf24l01 module with PA+LNA the 3.3V output of your Uno might not be able to provide enough current. In this case you can try a lower RF output power.

alexkwinto, 2017-02-20:
no im powering nrf24l01 with separated 3.3v bec. i checked the operating voltage -3.29 so shoud be fine.  also found some trolette action when the thorelette at minimum, if you need i can make a video to show.

goebish, 2017-02-20:
Maybe the problem is your PPM output then, have you checked its voltage (with a scope, this can't be done with a simple multimeter ...)

alexkwinto, 2017-02-20:
i dont have a scope to check it, the only way on my mind to debug it is to print throlette values to the port monitor of arduino IDE.

goebish, 2017-02-20:
Yes you can try that, but try to display the value only once per second for example, because I think using the UART too much will mess with PPM interrupt.

alexkwinto, 2017-02-20:
can you please advice how to do that? im not that good with time delayed tasks. also what is a good place in code to set prints?

goebish, 2017-02-20:
put that at the end of update_ppm:

static uint32_t next=0;
if(millis() > next) {
    next = millis()+1000;
    println(ppm[THROTTLE]);
}

alexkwinto, 2017-02-20:
yes the numbers are not stable:

1252
1186
1414
1199
1415
1187
1414
1414
1011
1000
1413
1000
1414
1414
1414

goebish, 2017-02-20:
Which transmitter are you using ? You need to adapt the PPM output.

alexkwinto, 2017-02-20:
AT9

goebish, 2017-02-20:
Do you have a n-mosfet handy ?

alexkwinto, 2017-02-20:
yes STN4NF20L, it is supplyed with 3.3 BEC througth the 10k resistors on vcc and ppm

goebish, 2017-02-20:
Here's a schematics, it should work with virtually any small power n-channel mosfet, such as the ones used to drive the motors in small toy quads.
http://i.imgur.com/Km99vxX.png

If you do not have a n-mosfet you can try to make a voltage divider with 2 resistors, but I can't recommend a ratio for the 2 resistor as I've no idea of the voltage of your PPM output. The n-mosfet solution is more universal.

goebish, 2017-02-20:
the 4NF20L should be fine, its Gate threshold voltage is low enough.

alexkwinto, 2017-02-20:
also the ppm signal is stable as i checked it with simulator flights

goebish, 2017-02-20:
On some transmitters, the PPM output is open drain, in this case, try to add a 10K resistor between PPM output and +3.3V to pull it up.

alexkwinto, 2017-02-20:
if i use your print it looks fine but if i print more often sometimes i can see that:

1373
1000
1000
1000
1373
1373
1373
1373
1365
1365
1365
1374
1374
1374
1374
1374

goebish, 2017-02-20:
Have you tried to add a pullup resistor on the PPM output (transmitter side) ?

alexkwinto, 2017-02-20:
i added it only ppm-3v3

goebish, 2017-02-20:
yes, also add one on the fet's gate, 
3.3V ------[10K]------Gate

alexkwinto, 2017-02-20:
as i said it was done

goebish, 2017-02-20:
No idea then, you might ask to this guy, he got the module to work with his AT9:
https://github.com/Sulya/nrf24_multipro_Radiolink

alexkwinto, 2017-02-20:
ok still a BIG thank you! offtopic: do you know were i can take a look at h8 mini scematics? seems one of the motor leading transistors is dead.

goebish, 2017-02-20:
Just replace the fet that's next to the non working (or always on) motor,  IRLML2502 is a good candidate, better replace all 4 so they're the same.
You can find some H8 mini experts here:  https://www.rcgroups.com/forums/showthread.php?2512604-Eachine-H8-mini-acro-firmware ;)

alexkwinto, 2017-02-20:
thank you again. an other offtopic question - have you met the issues with programming arduino pro mini?

goebish, 2017-02-20:
I've programmed hundreds of pro-mini without issues.

alexkwinto, 2017-02-20:
my is not programming with FTDI or arduino isp

goebish, 2017-02-20:
Maybe it's defective, I get mines in batches of 50, sometime I find a few duds, this hasn't happened for months tho.

alexkwinto, 2017-02-20:
i think in that case i will just replace it with atmega328 and external quartz

goebish, 2017-02-20:
Yeah you can do that, I like the pro-mini, it's small, easy to work with, and it doesn't cost more (sometime even less!) than a bare chip + crystal.