goebish/nrf24_multipro #26: Eachine E010 using different protocols?

mgrapiniak, 2017-02-20:
Hi,

Has Eachine released different protocols for their E010 product?
I do have a cheap Banggood product which I have reflashed using current version of your firmware (I know it is unsupported completely and understand why) but before purchasing your solution I'd like to know if that is the case.
Suprisingly Binding to cx-10 works as expected but no go with E010. Has anyone cone across this issue with your module?

goebish, 2017-02-20:
There's only 1 E010 protocol, but this protocol requires an accurate oscillator circuitry, which the BG module does not have. My modules are guarantied to work ;)

mgrapiniak, 2017-02-20:
Thanks for the info. 
Well this is what you get if you buy cheap stuff right?;)
Feel free to close out the ticket.

Cheers!

closed 2017-02-20 by mgrapiniak