goebish/nrf24_multipro #29: Adding support for electric skateboard

jacobbloy, 2017-06-30:
Is there any way that support could be added for an electric skateboard? The evolve gt uses nrf24lo1+ and seems to operate the same as an RC transmitter, it even has telemetry.
Because of the cheap way it was made I wouldn't be supprised if It is using an existing protocol.

Would you need both the board and transmitter to reverse engineer or could you do with just the transmitter? The transmitter has a cheap stc15w4k32s4 chip on it which I think I have seen on some china transmitters. Jacob.bloy@gmail.com

goebish, 2017-06-30:
Hi, if it has telemetry that's a 2-way protocol, I'd need both the
transmitter and the receiver.

/ Goebish

shusain, 2017-06-30:
https://github.com/shusain/eskatecontroller
A buddy just shared this issue with me I have a repo specifically for those modules and eskateboard control, it's very simple but is an easy springboard point.  I have an order of pcbs coming in today from China let me know if you're interested.

jacobbloy, 2017-06-30:
Thanks for the repo link, the evolve gt doesn't use a pwm input as far as I know, it uses SPI direct to the nrf module, the reason for doing the reverse engineer is to make a new remote using better developed code or replace the devolved esc with a vesc sink keep the remote.

@goebish could you email me your location and some information and I will get a board and remote to u while your working on this and we will work some thing out for your help!

jacobbloy, 2017-06-30:
[image]
[image]

shusain, 2017-07-01:
Gotcha yeah just made the PCB to connect the nrf and used an arduino pro mini 3.3V on as the transmitter and another as the receiver. I use the VESC so wasn't really worried about support for other signal protocols.

Nice looking board layout and controller setup so far, I've thought of putting an OLED in mine too but only have a b/w one and only .96" still might be worth it just for some basic Voltage readouts but there are good apps already (metr) for monitoring data the VESC collects (V, amps, temperature, etc.)