goebish/nrf24_multipro #30: flysky i6

irsaadinugraha, 2017-08-18:
how to change the protocol for flysky i6...
need the ino file

goebish, 2017-08-18:
Hi, there's nothing particular to do for the i6.
Just set channel order to TAER on the transmitter, or change order to AETR in the code.

irsaadinugraha, 2017-08-18:
ok i will try thanks

irsaadinugraha, 2017-08-19:
i try to change the code to AETR but not work...i still learning the command...can you make the code for flysky with AETR

goebish, 2017-08-19:
You only have to move THROTTLE, just after ELEVATOR, in nrf24_multipro.ino, that's all ...

AILERON,
ELEVATOR,
THROTTLE,
RUDDER

irsaadinugraha, 2017-08-19:
yes i do that but...but when finish i try to bind to eachine e011 still not bind...the light of arduino work find when binding...but not binding to my eachine e011

irsaadinugraha, 2017-08-19:
i'm using nrf24l01+PA and arduino mini pro..

goebish, 2017-08-19:
Check your wiring, nothing special is required in the code for the i6, many people got the module to work with it already.
http://carrerasfpv.com/forums/topic/2986/modulo-goebish-para-volar-los-minis-en-fly-sky-fs-i6/view/post_id/19103

irsaadinugraha, 2017-08-19:
ok thanks

noer96, 2018-01-27:
What is work to futaba t6j?

goebish, 2018-01-27:
Yes, it works with Futaba transmitters.

noer96, 2018-01-29:
Maybe I need you Help,  because I beginer for coding