goebish/nrf24_multipro #33: Quadbox 1602 support

deadmoo83, 2017-09-09:
Does this drone already work with an existing module? If not, what tools do you use to reverse the protocols? I was planning on setting up a nrf24L01 as a sniffer with an arduino or a raspberry pi adn using one of these projects, https://github.com/Yveaux/NRF24_Sniffer or https://github.com/yetifrisstlama/nRF24L01-sniffer to collect packets.

goebish, 2017-09-09:
I use a logic analyzer when possible, that's the simplest method.
https://www.deviationtx.com/forum/protocol-development/4595-spi-capture-guide

deadmoo83, 2017-09-09:
That's great. I have a generic to Cypress FX2 based logic analyzer that I rarely get to use. It works with Sigrock Pulseview. Procedure should be the same even though my hardware and software is different. This is going to be fun.

deadmoo83, 2017-09-09:
BTW, if you didn't know, Sigrock has a NRF24L01 decoder that stacks on the SPI decoder. Looks like it works with most NRF24L01 compatibles like the XN297 too.

deadmoo83, 2017-09-09:
I finally open the transmitter and it looks like it using a HS6200, which you had to use SDR to work with E012. So it doesn't look it will work with NRF24L01. What is the major difference that makes makes these chips on-the-air incompatible?

goebish, 2017-09-09:
I'm not sure why they're not compatible, they're supposed to use the same center frequencies and gfsk deviation but I wasn't able to make it work properly with a nrf24l01 as Tx but it works perfectly with a SDR.
Maybe there's one thing I haven't tried yet, I've to try something when I get a chance, just follow the e012 thread on the deviation forum ;)