goebish/nrf24_multipro #34: MJX X101 protocol

Kobbe1, 2017-09-27:
Hi,
awesome project that is!

I have an older X101 drone sitting around, with the remote boken.
As the spare reotes state they can be used for X101 and X600 I thought I try wether it works with your code.
It does not seem to bind - no reaction of the drone, just blinks away. I am assuming the arduino/NRF will autobind and I dont need to move the left stick up-down  like on the orignal remote.
Any ideas? Should it work?

Best,

-T

goebish, 2017-09-28:
If it's using the x600 protocol then it should work. Are you sure you're
selecting the right protocol and the ppm input is correct ?

Kobbe1, 2017-09-28:
Well, I have used a serial interface as on the fork from https://github.com/perrytsao/nrf24_cx10_pc., I updated it with the MJX code from your repo.
I have a couple of serial messages so I am sure it correctly calls MJX_init() and MJX_bind().

On the real remote you got to move the throttle stick full range and back to 0 to bind but I am assuming this is part of the MJX_bind().

The drone might as well be broken of course ...

silver13, 2017-09-28:
Can you try the code with anything else? The nrf24 can have issues if a cap is not used on it's power supply

Kobbe1, 2017-09-29:
I changed the power supply and soldered a cap on the module - just in case - no effect.
I tried with a FQ777-124 [(pic)](https://photos.google.com/share/AF1QipOh2D01ou-SOgV2lgG3-9fhTicV6PViQbstBMf8eo-nLzXerkpbp4DeJ1qcrtGAaQ?key=eU44N2tmT3ZVam9BRzZiRXNIMFFnN0x3aXBZUGtR) I have - does not bind.
I swapped the module - no effect.
I got rid of perrytsao's fork and used yours directly, just commented out the ppm parts and hardcoded the protocol selection - no effect.
The module works as a wifi scanner - but no luck with any of the 2 drones (FQ777-124 or MJX X101) so far.

silver13, 2017-09-30:
did you connect ce pin to nrf24 vcc?

Kobbe1, 2017-09-30:
It was on Arduino pin 7, tried to connect to VCC - no luck. Any ideas?

goebish, 2017-10-01:
CE is software driven (it might be useful later if we must do auto ack with xn297 emulation) so that's not your issue.

Kobbe1, 2017-10-03:
Here is what I have done:
Added another NRF24 on another Arduino (ProMini) - same behavior.
Tried the 'getting started' example of the RF24 library - works in both directions.

Could the X101 maybe just be broken? Or the protocol not be compatible? The  FQ777-124 did not work either however ...

Kobbe1, 2017-10-03:
I may have found something: NRF24L01_Reset() returns 0. Looking at the function in nRF24L01.ino it seems both status1 and status2 are 0 in NRF24L01_Reset().

silver13, 2017-10-03:
try reading the address register to make sure it's communicating correctly

silver13, 2017-10-03:
if you have an android mobile with ble you could also use that to check function

here is a "bayang ino" file which sends a ble signal
https://pastebin.com/hBgS6jLA

on the phone you can use the "nrf connect" app to see what is received

Kobbe1, 2017-10-10:
Tried the app, it can "see" the nRF024.
I used the link below from your bayang ino as this did not compile. What now? I read in another issue that NRF24L01_Reset() is not supposed to return 0 - but I don't know what these registers do or what it should return ...

 https://github.com/lijunhw/nRF24_BLE/blob/master/Arduino/nRF24_BLE_advertizer_demo/nRF24_BLE_advertizer_demo.ino

goebish, 2017-10-10:
NRF24L01_Reset() should return 1, if it returns 0 that means SPI is not properly connected, check your pinout / wiring.
https://github.com/goebish/nrf24_multipro/blob/master/nRF24_multipro/nRF24_multipro.ino#L39

Kobbe1, 2017-10-10:
I'll be damned, I can connect now!
I had changed the pinout in your sketch as it was more convienent to connect, but now changed it back to the defaults as you might be using these specific pins for a reason. Stupid mistake - thanks for the help!

closed 2017-10-10 by Kobbe1

goebish, 2017-10-10:
If you change the pinout you also have to edit the direct port access masks below.