goebish/nrf24_multipro #35: Eachine E013 support

etulupov, 2017-10-05:
Is it possible to capture and implement new protocol please? It's new version of E010 quad, but now the protocol is not compatible.

goebish, 2017-10-05:
Check my answer on the DeviationTX forums:
https://www.deviationtx.com/forum/protocol-development/7324-eachine-e013-small-pepper-pretty-please#64843

EZerkal, 2019-05-02:
Are there any decisions for this problem?

Lethalinjectionx, 2019-05-09:
I was trying to find a resource to bind this little quad to my Jumper T16. I recently got an eachine e013 and the TX has this chip in it.
http://www.essemi.com/product/66.html
translated into english http://www.translatetheweb.com/?from=&to=en&dl=en&a=http%3A%2F%2Fwww.essemi.com%2Fproduct%2F66.html

I found the following spec documents and I translated them from Chinese to English.
[HW2171B_Datasheet_C V1.2.zh-CN.en.pdf](https://github.com/goebish/nrf24_multipro/files/3163518/HW2171B_Datasheet_C.V1.2.zh-CN.en.pdf)
[AN1029_HW2171B_Hardware_Reference_Design V1.1.zh-CN.en.pdf](https://github.com/goebish/nrf24_multipro/files/3163523/AN1029_HW2171B_Hardware_Reference_Design.V1.1.zh-CN.en.pdf)

I believe the RX is this chip which also lists documents for download.
http://www.translatoruser-int.com/translate?&to=en&csId=133ad314-0f67-4418-9c27-2b1d92a767b7&usId=6394414a-db14-4835-b0e1-63a761a66c6f&dl=en&ac=true&dt=2019%2f5%2f9%2018%3a45&h=Klb6DcTsQWKEuF5O_Ub-rPiYib48r7oE&a=http%3a%2f%2fwww.essemi.com%2fproduct%2f65.html