goebish/nrf24_multipro #36: Telemetry beep alamrs

grezlo, 2017-10-24:
Hello! Thanks for a great product! 
I want tuning it on this way: add a beeper 5V, and to inform them about the battery and RSSI.

219 telemetry_data.volt1 = chanval.value; 

What is the range of values for this parameter?

goebish, 2017-10-24:
If I remember well it's vbat*10, ie 42 for 4.2V, but this has to be checked.

grezlo, 2017-10-24:
You wrote:
        if (packet[0] == 0x85 && packet[14] == Bayang_checksum()) {
            // uncompensated battery volts*100
            chanval.bytes.msb = packet[3] & 0x7;
            chanval.bytes.lsb = packet[4] & 0xff;
            telemetry_data.volt1 = chanval.value;
So vbat*100 correct? Thank you!

And RSSI is 0-100 parameter correct?:
            telemetry_data.rssi = Bayang_telemetry_count * 2;

goebish, 2017-10-24:
I don't remember, just try and add some serial logging if required ;)

grezlo, 2017-10-24:
Well I will write you about the results. Thank you!

goebish, 2017-10-24:
Also, keep in mind telemetry range is very short as the H8 mini, B03 and other "Silverizable" quads do not have a PA.

grezlo, 2017-10-24:
What about E011?

goebish, 2017-10-24:
Same as B03, none of the supported model has a PA onboard.

grezlo, 2017-10-24:
There is no possibility to add an amplifier? :) like PA LNA Wifi modules

goebish, 2017-10-24:
No.

grezlo, 2017-10-24:
Ok. Will short telemetry :/ 100 meters realy?

goebish, 2017-10-24:
Way less than 100m ... even indoor, telemetry stops working 1 or 2 rooms away. But that's not a big deal, you still have control ...

grezlo, 2017-10-24:
For check the VBAT need flying closer to myself)

goebish, 2017-10-24:
Yes, that's the only way ;)

grezlo, 2017-10-24:
Thanks again. Will try.

silver13, 2017-10-25:
rssi is about 200 normal operation, so not 0 -100