goebish/nrf24_multipro #38: Bayang doesnt bind

Wene001, 2017-10-31:
Hi Goebish
Thanks for a great peace of software
For testing i have flashed your code to a arduino uno and connected a NRF24 from Bangood.
Controlling an E010 from my Taranis works fine, but i couldnt bind to Bayang toys.
Tested with silverized BwhoobB03pro and original Eachine E011.
Any ideas?
Thanks from Austria
Wene

goebish, 2017-10-31:
No idea, Bayang protocol is working fine here.
Try to transfer this model to your Taranis (OpenTX 2.2) using the Companion:
https://www.dropbox.com/s/e9oqo5ye6u0mke9/H8_SilverX.otx?dl=1
(no need to move sticks at startup with this model)

Wene001, 2017-11-01:
Thanks for your help
Arduino UNO is powered over USB for testing and TX is switched on first.
No way to bind at Power up. Binding via Switch @AUX8 works fine.
If i try to bind at Powerup i have to disconnect/connect lipo on the quad, to be able to bind via AUX8 switch.
Binding to E010 works at Powerup and with Switch.
?!?

Wene001, 2017-11-01:
Problem solved.
This happens only when the arduino power over computers USB.
If powered e.g. from a USB Powerbank binding works perfekt.

Next step make a beeper working for Bayang telemetry undervoltage and transplant the whole thing in a cheap Walkera transmitter...

closed 2017-11-01 by Wene001