goebish/nrf24_multipro #40: New board for CX-10

soulman301, 2018-03-23:
It looks like the manufacturer recently updated the electronics. The transmitter is still 2.4 GHz, but it seems the connection protocol has changed. Any advice?

samnockels, 2018-06-07:
I would also like some advice with reverse engineering this new protocol. 
Here are some pic of the new electronics -> https://imgur.com/a/vCBHihA

goebish, 2018-06-07:
That's not a good sign, there's only 1 chip in the Tx ... (MCU+RF in the same package), perhaps it's xn297 compatible, you'd need a nrf24l01 sniffer (set 2 byte address to 0x0f,0x71 to try to detect xn297 packets) or an SDR with 2.4GHz capability to check that. If it's xn297 compatible then it can be emulated with the nrf24-multipro. If it's based off LT8910 or something else there's nothing much we can do for now ...

soulman301, 2018-06-07:
There's no guarantee they still have them, but as of mid April Walmart.com was still selling the old version

samnockels, 2018-06-07:
I'm from the UK and can't seem to get hold of one, bought 5 from different sources and all are the new one :(  @goebish I'm a newbie in rf, how would I make a sniffer using the nrf24lo1?

goebish, 2018-06-07:
Send me one if you want (my nickname @gmail.com for address, I'm from France) but I can't promise I'll be able to make it work, if it's using the same kind of transceiver than the F36 there's nothing I can do.

samnockels, 2018-06-07:
Just sent an email :)

goebish, 2018-06-07:
I didn't receive it (not in spam folder either), I'm going to race with friends now, I'll check my emails tonight ;)

goebish, 2018-06-07:
@soulman301 does yours also have only 1 IC in the TX ?
(and 2 in the quad, [MCU+RF] + MPU).

samnockels, 2018-06-07:
@goebish What other drones do you know of that still work?

goebish, 2018-06-07:
You mean from the compatibility list on the project page ?
All of them except a few CX10 ;)

samnockels, 2018-06-08:
Cool, I'll have a look at them. I won't be able to send you the electronics until next week, how would you go about sniffing out xn297 packets? I might as well have a go at it this weekend before I send you it on Monday.

goebish, 2018-06-08:
I would set the nrf24 address length to the (invalid ;)) value of 0, set address to 0x0f,0x71 (part of xn297 preamble), set it to receive mode, then listen to every frequencies with 250kbps and 1Mbps bitrate.
If similar packets are received that's a good sign, then they've to be unscrambled ...

goebish, 2018-06-08:
here's a good article if you're interested in sniffing nrf24 packets:
http://travisgoodspeed.blogspot.com/2011/02/promiscuity-is-nrf24l01s-duty.html
xn297 is easy to sniff with a nrf24l01 because it always has the same preamble (0xf7155)

kpfaulkner, 2018-07-11:
Just got myself a CX10... but no luck running this. Guessing I've got a "new one" as well :(  Can I do anything to help potentially get this fixed?

kpfaulkner, 2018-07-12:
Just took apart the one I recently got. Similar images to the one shown above, but on the FC it has the marking KY-CX10R-6...

goebish, 2018-07-12:
If you've an HackRF or any 2.4GHz capable SDR then send me some captures, or send me one of those new CX10 (without guaranteed results), that's the best I can tell you ;)

kpfaulkner, 2018-07-12:
Found out a local shop has a JJRC H36 so might bite the bullet and just switch to that for now :)  Hopefully I wont have the same issue where they've changed the internals :/   Whats the HackRC? Any links?

goebish, 2018-07-12:
You mean HackRF ?
https://greatscottgadgets.com/hackrf/

Take care with the E010 / H36, they require to use a nrf24l01 module with an accurate crystal oscillator, those ones work fine: https://www.banggood.com/2_4G-NRF24L01-PA-LNA-Wireless-Module-1632mm-Without-Antenna-p-922601.html

goebish, 2018-07-12:
But honestly, the E010 and H36 (same machine actually) are underpowered, get a Boldclash B03 pro (get the pro, you don't want the version with baro, it's not fun and it's more expensive) or an EAchine E011, you won't regret it.

kpfaulkner, 2018-07-12:
From a flying point of view my son and I fly 5" racing quads and occasionally tiny whoops. These smaller quads are purely to satisfy my curiosity about starting to code one. 

As long as I can arm it, programmatically hover and move about a little (again just to satisfy my curiosity) that would probably be enough (for now). 

I'd love to see how far this can go and  decide flysky protocols etc. (ever looked at that?)

This is a great GitHub repo for learning!

Cheers

goebish, 2018-07-12:
YMMV but most of the modules you linked don't have accurate enough oscillator for the E010 protocol (because of the xn297 emulation @ 250kbps bitrate) and they don't have a power amplifier, so even if it works, the range will be very short. At least you can keep the small power supply board, the board I linked is pin compatible ;)

Flysky protocols require an A7105 transceiver, not a nrf24l01, I reverse engineered the AFHDS 2A protocol some time ago: https://www.deviationtx.com/forum/protocol-development/5251-flysky-afhds-2a-protocol-as-used-i10-i6-it4

kpfaulkner, 2018-07-12:
Any recommendation for the A7105 transeiver? (sorry for hijacking this issue/thread)