goebish/nrf24_multipro #42: NiHui NH-010R

kpfaulkner, 2018-07-13:
Attempting to get this working with (what I believe is a NiHui NH-010, but branded slightly differently).
Have attached an image of the board. It shows JR-NH010R on the FC.

I *believe* it can bind....  at least the light stop flashing (similar to when using the controller) when the bind function completes.

I'd just like to check if this particular FC is known to work with this project? (if not, I might be able to return to the store).

kpfaulkner, 2018-07-13:
[image]

goebish, 2018-07-13:
Seems that's an E010 / H36 board, it should work if you've a nrf24l01 board with an accurate enough crystal oscillator.

tinywhoopdestroyer, 2024-02-12:
Hi, I want to ask that is there anyone know that the H36F board (JR-NH010R7) is supposed to be Rev7 right? 
And I have heard that account Goebish is an expert in these binding and analyzing the code of these rc drone protocol, so could you (Goebish ) tell me that can the Tx FS-I6 with IRange Irx6 bind with the H36F circuit board ? Because in recent years i have seen that the Eachine E010 new supplies (Rev5, red circuit board ) can not anymore bind with the Tx I6 + irx6 (which both E010 and H36, H36F used to be bind which Module NRF24L01, protocol MJxq, sub-protocol E010) . Due to lack of money so i can not pay for 4in1 module Tx . I really appreciate any of your guidance 
[image]
[image]

ahmadazmi19, 2024-05-15:
I also have an FC JR-NH010R but in the past when the + and - poles of the battery were installed were reversed, can the FC be repaired again?

goebish, 2024-05-15:
Since there's no protection circuit (diode), I don't think it can be repaired, all the ICs are probably toast.

ahmadazmi19, 2024-05-15:
for FC JJRC H36 using FG JR-010 R8 mode in which altitude hold can be changed to manual mode? because I want to connect to my manual remote jumper but I can't because the FC mode is altitude hold....

ahmadazmi19, 2024-05-15:
For FC JJRC H36 using FC JR-010R8 with the default altitude hold mode, can it be changed to manual mode? because I want to connect to my manual remote jumper but I can't because the FC mode is altitude hold....

goebish, 2024-05-15:
No idea, I've been out of the loop for quite a while ... But if altitude hold can't be switched off with the stock controller then that's how the FC firmware works, you won't be able to switch it off.