goebish/nrf24_multipro #43: JJRC H36

markoni985, 2018-07-25:
Hi, I tried to use this project to control my JJRC H36.
I tried to force binding to it by making a change to  nRF24_multipro.ino.
Function selectProtocol() was edited to set current_protocol = PROTO_E010;
When I power up Arduino it does not bind to a drone.
Did anybody successfully bind to JJRC H36 using this project?

goebish, 2018-07-25:
Do you have a PPM input ?
The protocol won't start unless safe throttle position is detected.
Also, the E010/H36 protocol requires to use a nrf24l01 module with an accurate crystal oscillator, such as this one: https://www.banggood.com/DEVO-Transmitter-Tuner-Modified-for-FBL80-FBL100-WLtoys-V922-p-89812.html
Most of the cheaper nrf24l01 modules have issues with this protocol.

markoni985, 2018-07-25:
Thank you for the fast reply, I was using a cheap nrf24l01 module. That might be the biggest contributor to the issue. The PPM input is not present. Also, I am trying to replicate what https://github.com/perrytsao/nrf24_cx10_pc did, But his project does not have all the protocols and files from your project.

goebish, 2018-07-25:
cx10 are less sensitive to the nrf24 module quality as they're using 1Mbps bitrate, but the xn297 emulation @ 250kbps (required by the e010/f36) requires a "good" module.

markoni985, 2018-09-15:
Hi "goebish" , I just got nrf24 that you suggested and it works great.
Another interesting fact after it connected to the drone I tried old module back again and the old cheap module now connects from time to time too.
Anyway thank you!

goebish, 2018-09-15:
Thanks for reporting, closing this issue.

closed 2018-09-15 by goebish