goebish/nrf24_multipro #44: Doesn't compile in Arduino 1.8.15 (not sure about other versions)

voroshkov, 2018-10-27:
I download archive and open nRF24_multipro.ino --> Arduino IDE opens.
Click compile icon in Arduino IDE  --> get multiple errors "xxx was not declared in this scope"

voroshkov, 2018-10-27:
Thanks for the project anyway! :)

ZeliangSu, 2018-12-16:
I have the same issue on Mac OS Mojave.

ZeliangSu, 2018-12-16:
@voroshkov I got the same problem but I figured it out. In my case, the code sketch exploded the memory of the Arduino. I deleted all the comments and it works like a charm. Hope it's useful for you.

SteveQUT, 2018-12-16:
Can confirm also have this issue.

goebish, 2018-12-16:
Seems like the last Arduino IDE doesn't appreciate long comments at the beginning of the main file.
Download / checkout the project again, it should be fixed now.

closed 2019-01-22 by goebish