goebish/nrf24_multipro #46: Eachine E016H support

BLeibbrand, 2019-02-06:
Hi all,
Thanks for the wonderfull work.
I love flying with my Taranis X9D+ and IRX4+ TX Module in combination with an Eachine 010.

I bougth an Eachine E016H which looks similar to the E010 but with more prop protection. 
I have kids and dogs so protection is needed.
But sofar I'm not ably to get the combination working.
Which protocol do need to use, or do you have to reverse engineer it first?

OpenTX Companion - 2.2.3
Taranis  FW:opentx-x9d+ version 2.2.3 (ae5ac05b)
IRX4+ - original (no flashing yet. Need to get USB - FTDI connector)

Eachine E016H
https://www.banggood.com/Eachine-E016H-Mini-Altitude-Hold-Headless-Mode-8mins-Flight-Time-2_4G-RC-Drone-quadcopter-RTF-p-1379560.html
[image]

goebish, 2019-02-06:
If it doesn't work with any (sub)protocols with DeviationTX or MultiModule then it needs to be reverse engineered. Can you open the transmitter and quad and tell me which ICs they're using ?

BLeibbrand, 2019-02-06:
Hello Goebish,
 
Thanks for the quick response.
I did some home work.
I hope this helps.
 
Yours Sincerely, 
 
 
Bert Leibbrand                                      <><
M +31 (0)6 27 74 60 88
 <mailto:bert.leibbrand@itri.nl> bert.leibbrand@itri.nl 
 
 
Met vriendelijke groet, 
 
 
Bert Leibbrand                                      <>< 
M +31 (0)6 27 74 60 88
 <mailto:bert.leibbrand@itri.nl> bert.leibbrand@itri.nl

goebish, 2019-02-07:
I see nothing. If you want to post pictures I suppose you've to use github, you can't insert them in an email.

BLeibbrand, 2019-02-07:
[PAN163CX  Productspecificatie.docx](https://github.com/goebish/nrf24_multipro/files/2839702/PAN163CX.Productspecificatie.docx)
[PAN163CX  Productspecificatie.pdf](https://github.com/goebish/nrf24_multipro/files/2839703/PAN163CX.Productspecificatie.pdf)
[SchematicsforESP32.pdf](https://github.com/goebish/nrf24_multipro/files/2839704/SchematicsforESP32.pdf)
[image]
[image]

BLeibbrand, 2019-02-07:
[image]
[image]

goebish, 2019-02-07:
Ok, that's a PAN163CX, it can be reverse engineered, but not with a logic analyzer.
If someone dropships one to me I can take a look.

BLeibbrand, 2019-02-09:
Hello Goebish,
I just had time to unscrew the transmitter.
The transmitter pictures.
The chip: XNS1042CV
[image]
[image]
[image]
[image]
[image]
extra info about transmitters with the XNS chip:
https://www.google.com/imgres?imgurl=http%3A%2F%2Fimg.over-blog-kiwi.com%2F0%2F55%2F43%2F29%2F20181129%2Fob_463a88_xwqz238505.jpg&imgrefurl=http%3A%2F%2Fchezcyril.over-blog.com%2F2018%2F11%2Fmini-drone-pliant-volcano-w606-16-un-drone-qui-se-pilote-grace-a-un-gant-et-au-mouvement-de-la-main-gyroscope.html&docid=ztPRrdg6XvaYsM&tbnid=nXG7m4h00eZ-eM%3A&vet=10ahUKEwjnhpD3467gAhWCPFAKHdPHDGEQMwgoKAAwAA..i&w=1205&h=544&bih=642&biw=1396&q=XNS%201042CV&ved=0ahUKEwjnhpD3467gAhWCPFAKHdPHDGEQMwgoKAAwAA&iact=mrc&uact=8

http://chezcyril.over-blog.com/2018/11/mini-drone-pliant-volcano-w606-16-un-drone-qui-se-pilote-grace-a-un-gant-et-au-mouvement-de-la-main-gyroscope.html

goebish, 2019-02-09:
As I said, I believe it can be reverse engineered but I don't have one. If someone wants to dropship one to me send me an email @gmail.com

goebish, 2019-02-27:
https://www.youtube.com/watch?v=GXEfK9D_90Y
😜

BLeibbrand, 2019-02-27:
Great !!
Which steps do I have to take to get it into my Taranis / iRange?
 
Met vriendelijke groet, 
 
 
Bert Leibbrand                                      <>< 
M +31 (0)6 27 74 60 88
 <mailto:bert.leibbrand@itri.nl> bert.leibbrand@itri.nl

 
https://www.youtube.com/watch?v=GXEfK9D_90Y
😜

goebish, 2019-02-27:
I've to cleanup my code then I'll contact @pascallanger so he can add it to the 4-in-1 module firmware.

BLeibbrand, 2019-02-27:
What do you think of the indoor capabilities?
 
Met vriendelijke groet, 
 
 
Bert Leibbrand                                      <>< 
M +31 (0)6 27 74 60 88
 <mailto:bert.leibbrand@itri.nl> bert.leibbrand@itri.nl

goebish, 2019-02-27:
It's fun but quickly boring ;)
It should be perfect for a beginner or a kid as throttle is easy to manage.

BLeibbrand, 2019-02-27:
Did you also enable the three fly modes?
 
Met vriendelijke groet, 
 
 
Bert Leibbrand                                      <>< 
M +31 (0)6 27 74 60 88
 <mailto:bert.leibbrand@itri.nl> bert.leibbrand@itri.nl

goebish, 2019-02-27:
Fly mode is always set to "expert" then you can set lower endpoints on your transmitter if required.

pascallanger, 2019-02-28:
New Multiprotocol version 1.2.1.26 available on GitHub: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module

Thanks to Goebish the E016H is now supported:
Protocol E01X (45)
Subprotocol E016H (2)
Channels:
CH5=STOP
CH6=FLIP
CH8=HEADLESS
CH9=RTH
Please test and report.

Pascal

goebish, 2019-02-28:
Thank you!
@BLeibbrand please report once you've tested.

goebish, 2019-02-28:
@pascallanger I think there's a mistake here:
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/04ed613ca3bd4bcda53457504adeca7aa9053168/Multiprotocol/E01X_nrf24l01.ino#L269
it should be XN297_SetTXAddr for E016H and HS6200_SetTXAddr for E012 & E015

pascallanger, 2019-02-28:
Thanks! copy and paste issue... Corrected in v1.2.1.27.

BLeibbrand, 2019-03-02:
Hello Goebish and Pascal,
 
I have soldered the connectors in my 4in1 and have to wait untill Monday to pickup my USB-TTL to get the bootloader installed.
As soon as I have tested it I will inform you.
 
Yours Sincerely, 
 
 
Bert Leibbrand                                      <><
M +31 (0)6 27 74 60 88
 <mailto:bert.leibbrand@itri.nl> bert.leibbrand@itri.nl

BLeibbrand, 2019-03-04:
The ST-LINK V2 arrived today. No luck in connecting to the 4in1.
Any help for the cable connection would be fine.
[image]
And 4in1 GND, TX, RX

pascallanger, 2019-03-04:
Why a st-link? you need a FTDI like the one in the documentation: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Compiling_STM32.md#tools-required
Have you tried to plug the USB to your PC? The latest modules are shipped with the bootloader flashed.

goebish, 2019-03-04:
The STLink connects to a SWD interface (swclk, swdio, gnd), if you want connect to UART (gnd, rx, tx) you need an USB to UART adapter (FTDI dongle) instead.
I'm not a 4-in-1 expert, you'd better ask on rcgroups https://www.rcgroups.com/forums/showthread.php?2165676-DIY-Multiprotocol-TX-Module/

BLeibbrand, 2019-03-04:
I already tried the USB: 
[image] 
It's a 4in1 from june 2018.
So I'll get a USB-UART and after that we can continu.

pascallanger, 2019-03-04:
Yep no bootloader installed. Make sure to purchase the right FTDI which can run at 3.3v as mentionned on the github doc.

BLeibbrand, 2019-04-01:
Hello guys, The FTDI finally arrived. 
I have loaded the version with the bootloader. 
The USB is now recognized as com25.  
OpenTX is version Companion is 2.2.3 (7 jan 2019)
[image]
I can only see the first two sub protocols.
[image]
In the Taranis:
Mode:                     MULT   E01X E012 or E015 (no E016)
Module Status        V1.2.1.29
What is missing / what should I do?

goebish, 2019-04-01:
Seems OpenTX Companion doesn't allow the selection of a custom protocol #
On the transmitter, select Custom protocol, with options 45 & 2:
[image]

BLeibbrand, 2019-04-01:
It still doesn't bind

goebish, 2019-04-01:
line 268 is wrong too:
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/04ed613ca3bd4bcda53457504adeca7aa9053168/Multiprotocol/E01X_nrf24l01.ino#L268
it should be:
if(sub_protocol!=E016H)

BLeibbrand, 2019-04-01:
[image]
Still only blinking drone.
I'm using Custom 45 & 2 
Bind and bind on power up

BLeibbrand, 2019-04-01:
I hope you have a next step to try.

goebish, 2019-04-01:
Sorry, revert the change on line 268, I wasn't looking at the last version of the file, it should be
if (sub_protocol == E016H)

goebish, 2019-04-01:
Your last error in the Arduino IDE is normal

BLeibbrand, 2019-04-01:
OK. Now it does bind.
If you give it about 65% throttle the rotors spin (smilar to the spin with the standard transmitter just after binding with the throttle UP and DOWN). (After some time it does sometimes take of but I'm sure why).
SF will disarm
SB will blink 2x

goebish, 2019-04-01:
I don't understand your last message, does it fly ?

BLeibbrand, 2019-04-01:
Left and Right seem to be opposite.
Yes, it does fly as soon as you move the throttle a bit down and up again like 80% to 70% and up again.
The spinning only starts above the 60-70% below that no motors.

goebish, 2019-04-01:
Ok that's normal.
You can revert Aileron and/or Rudder direction in the mixer.

BLeibbrand, 2019-04-01:
So the drone is flying. 
Attitude hold is good.
Drifting to the right. (trim complete to the left to make it stable)
Headless and Stop are working.
RTH seems to fly to the right.
No space for flip at this moment

goebish, 2019-04-01:
if it drifts put it on a level surface then set both sticks to bottom corners (I don't remember which ones) to recalibrate the accelerometer.
Or set a slight offset in the mixer for this channel.

pascallanger, 2019-04-02:
E016 should be fixed. Can you download the latest version from github and try?

BLeibbrand, 2019-04-02:
[image]
After reversing and trimming the drones is stable.

goebish, 2019-04-02:
Ok, I'll check if calibration requires an extra flag.
Is one key take off really required as pushing throttle just does the same ?

BLeibbrand, 2019-04-02:
One Key Take off would be fine to give new users an easy way to get the drone up without the up-down a bit - up  sequence needed at this moment. 
So I would say: Nice to have.

Calibration seems to be needed as the trim and the strange RTH direction might be caused by it.

BLeibbrand, 2019-04-02:
Can you see if this drone also supports other flight modes (like acro) because as you mentioned before. If you are able to fly a drone this might becomes boring when you have some flying space.

goebish, 2019-04-02:
One key take off would be a mess to implement, and not every controller has a momentary switch, honestly that's not a big deal if you've to push throttle instead of a button (this quad is *very* gentle ;))
Don't expect hidden features (acro mode...) on this kind of quad.
I found the calibration flag, I'm implementing it.

goebish, 2019-04-02:
@pascallanger added calibration flag:
https://github.com/DeviationTX/deviation/pull/844/files

BLeibbrand, 2019-04-02:
I have uploaded the newer version (1.2.1.30);
It is not recognized in the Taranis:
Mode: MULT E01X E012 or E015 (no E016)
So I still use the Custom 45 & 2
I have removed the reverses as they are fixed.
I only have to glue some components...
The rest is for tomorrow.

goebish, 2019-04-02:
That's normal, E016 will only appear in the menu with the next OpenTX update.

BLeibbrand, 2019-04-02:
Funny to see my other Eachine (E010 mini) is in an other group MJQX instead of E01X.
[image]

pascallanger, 2019-04-03:
Calibration available in the new version.

BLeibbrand, 2019-04-05:
V 1.2.1.31
BIND:
* Power on drone'
* Power on Taranis [Bind on powerup = checked] 
No binding. Menu => page => [bind] => binding.
(Can I use an other method for binding with this drone?)

Calibration:
After binding, both sticks to bottom outside corner: drone blinking about 5 times.
So calibration seems to work.

The weather was OK today, so I could also test the Flip.
If I have the flip-switch "On" the drone will flip even with a little stick movement.
Can I switch it On and Off before giving the flip command to ensure it will flip only once?

Thanks sofar.

pascallanger, 2019-04-05:
You need to learn to use your multi module and radio ;-)
 - multi offers an autobind option which if you have left the defaults in config.h enables you to bind with a switch assigned to channel 16. So basically, enable autobind in the model config and add a switch (temporary trainer switch is perfect) to channel 16. Each time you press it, it will launch a bind. Documentation: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#useful-notes-and-definitions
 - On your Taranis, you can configure a logical switch that will just send a pulse to the flip channel.
Pascal

BLeibbrand, 2019-04-06:
If found some info about three speed levels.

It is the first button in the manual (top left).
Is this already included?

https://drones-camera.com/eachine-e016h-mini-drone-review-in-depth/

------------------
Ease Of Use:
When it comes to drone use, it is pretty simple to control. Its control is fundamental that allows beginners to get hold of it very quickly. It has low battery alarm as well that gives you an indication about when to bring it back. **Multiple speed modes bring more comfort to the flight**. Most of the drones need smartphone app download to control, but in this one, things are pretty simple, you can control it with the controller only.
-----------------

goebish, 2019-04-06:
The protocol always use highest speed, if you need to tame it down just set lower rates (80%, 60%) in the mixer for aileron / elevator channel, as you would do with any hobby grade model.

Dgirdler, 2019-06-24:
Gentlemen.  I have the Eachine E016H and the Taranis X9D+.  Can you please direct me to where I can learn how to put V 1.2.1.31 (or the newest V) into this...Multiprotocol TX Module For Frsky X9D.  (https://www.banggood.com/Multiprotocol-TX-Module-For-Frsky-X9D-X9D-Plus-X12S-Flysky-TH9X-Transmitter-p-1097688.html?rmmds=search&cur_warehouse=CN)

goebish, 2019-06-24:
This project is outdated and I never supported this banggood module, if you want the E016H protocol, get a 4-in-1 Multi-Protocole Module.
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module

Dgirdler, 2019-06-24:
Thank you....learning....

Vr,
Dave

derFrickler, 2019-07-03:
Any info if the new Eachine E016F uses the same protocol?
https://www.banggood.com/Eachine-E016F-3-in-1-EPP-Flying-Air-Boat-Land-Driving-Mode-Detachable-One-Key-Return-RC-Quadcopter-RTF-p-1472961.html

Arkango, 2019-12-18:
Hello i found in a drone remote controller, an xns1042cv can be this discussion appropriate? That component implements the E016H protocol ?

goebish, 2019-12-19:
xns1042cv is a 8 bit mcu with built-in xn297 rf transceiver.
The E016H transmitter is using this IC but that doesn't mean that any other machine that's using the same chip is running the same protocol.

[14 more replies]

closed 2020-11-16 by goebish