goebish/nrf24_multipro #47: Bayang protocol Autobind

SirDomsen, 2019-02-10:
Hi again,

still experimenting with a Graupner MZ-24 and your module. I see everything works as intended, all 11 Bayang channels work, channel 12 hits reset. Issue is that once reset is hit or module is power cycled, it won't bind to the (Silverware) FC again. Is it possible that TX ID changes somehow? While using Deviation or toy TX, you can power cycle the TX and it will rebind to the (still powered) FC. 

Also, [Autobind](https://www.rcgroups.com/forums/showpost.php?p=39330532&postcount=2126) does not work, it seems like the FC sees the module as a different tx every time the module is power cycled...

Is there a way to get autobind to work properly?

cheers
Domsen

goebish, 2019-02-10:
Fixed.

SirDomsen, 2019-02-10:
Awesome, thanks!

SirDomsen, 2019-02-11:
Sorry for bugging you, but I just recognized something strange that I can't get sorted:
The MZ-24 uses the same channel mapping as Spektrum, so I commented in #define SPEKTRUM. So far so good, all 4 analogue channels (TAER) work as intended. Min and max values fit, middle also. I use a Silverized FC to debug, so I can actually see the channel values. Strange: All digital Bayang channels switch not at 50% ("0) as intended but at around 70% ("+40").
Another interesting thing is throttle channel: in Silverware it has a 10% safety travel. So My Devo starts to switch on the motors at -80 (Deviation). The Multipro does the same at -74. No big deal, just curious why...

goebish, 2019-02-11:
the #define SPEKTRUM is only for TXs such as the MLP4DSM with limited throw, you don't need it with the MZ-24.

goebish, 2019-02-11:
Mapping is TAER by default, same as Spektrum transmitters.

SirDomsen, 2019-02-11:
Ah, thanks, I didn't know. Now The analogue channels appear to be correct, digital channels switch at +50% though...

SirDomsen, 2019-02-11:
#define PPM_MIN 1100
#define PPM_SAFE_THROTTLE 1100 
#define PPM_MID 1500
#define PPM_MAX 1900
#define PPM_MIN_COMMAND 1300
#define PPM_MAX_COMMAND 1700

goebish, 2019-02-11:
PPM_MIN & PPM_MAX aren't meant to be modified, you should setup your controller endpoints to output 1000-2000µs

SirDomsen, 2019-02-11:
It does not matter strangely... Sure, if I set +- 125 i's ok with stock settings, but the digital channels still switch at around +40%. I set all channels to +-125%

goebish, 2019-02-11:
Ok that's weird, you can modify PPM_MAX_COMMAND if required, but not PPM_MIN & PPM_MAX

SirDomsen, 2019-02-11:
Got it sorted. 
#define PPM_MIN_COMMAND 1100
#define PPM_MAX_COMMAND 1500
did the trick. Thanks for your advise!

ANS-HAFEEZ, 2019-03-07:
Hey,
I am getting issue in binding with silverware. on powering on FC it first blinks fast. As, I turned on transmitter FC start blinking slowly but never get constant.
Any suggestion?

JamesKingdon, 2019-04-04:
Hi @goebish I just ran into this problem with an iRangeX iRX6. Probably an unfair question, but do you happen to know if the module is directly compatible with your original code? I'm wondering if I can flash with your latest and bring it up to date.

goebish, 2019-04-04:
No idea, sorry, I offer no support for those modules.