goebish/nrf24_multipro #49: Use of sticks position instead of PPM input for bayang protocol

ANS-HAFEEZ, 2019-03-07:
Hi,
Is it possible to change input from PPM to directly stick potentiometer.
Reading analog pins and map it to PPM value range.

goebish, 2019-03-07:
That's possible, read the comments under this video:
https://www.youtube.com/watch?v=bxsW8mlMUIg

goebish, 2019-03-07:
"First, remove (comment) the attachInterrupt function in setup() to disable the PPM input At the beginning of the selectProtocol function, set ppm_ok to true instead of false Then rewrite the update_ppm function, let's say the trhottle pot is connected to A2 (A0 & A1 are already used): ppm[THROTTLE] = map(analogRead(A2), 0, 1024, 1000, 2000); and so on for the 4 channels. If you want to add switches, in the setup() function declare some pins as INPUT_PULLUP eg: pinMode(6, INPUT_PULLUP); // aux1 switch connected between gnd & D6 then check it in the update_ppm function, eg: ppm[AUX1] = digitalRead(6) ? 1000 : 2000; // AUX1 = Channel 5﻿"

ANS-HAFEEZ, 2019-03-07:
Thanks.
I ll be back after testing

goebish, 2019-03-07:
also, if you only want to use teh Bayang protocol, in loop(), replace:
selectProtocol();
with
current_protocol = PROTO_BAYANG;

ANS-HAFEEZ, 2019-03-08:
It works Great.
Thanks for your work.

closed 2019-03-08 by goebish