goebish/nrf24_multipro #51: walkera 2401

maxvrpilot, 2019-08-04:
hi , i have one nrf24 for my jr TX , so please is possibile use it with walkera micro hely ? now  i use walkera 2401 tx for this walkera micro hely . regards

goebish, 2019-08-04:
that's not possible, Walkera protocols are using a CYRF6936 transceiver, not the NRF24L01

maxvrpilot, 2019-08-04:
thanks for the reply, can you recommend another method?

goebish, 2019-08-04:
4-in-1 Multi Protocol Module: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module

maxvrpilot, 2019-08-04:
hi dear than you again , I have a DSMX / DSM2 2.4GHz module I read in the link that you sent me that it is possible to load the walkera firmware on this module, do you know this procedure?

goebish, 2019-08-04:
That's probably possible with the OrangeTX module, procedure is here:
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Compiling_OrangeTx.md
I don't know how it works exactly, read the procedure on github first, then come here if you need further help:
https://www.rcgroups.com/forums/showthread.php?2165676-DIY-Multiprotocol-TX-Module

maxvrpilot, 2019-08-04:
you're great  thank you , now I read everything and I'll let you know. regards

maxvrpilot, 2019-08-23:
hi dear , i have received my jumper jp4in1 multprotocol module but i have tried to bind cx10 but nothing  so i have also tried with my walkera micro Hely ( i have 2401tx) and also nothing ! so please i need help about the DIAL selection wheel and I would like to know how do I recognize the direction of the wheel? I don't understand which DIAL I'm selecting! I also tried the procedure to turn off the green LED but I can't do this either and the green light is always on. My radio is a JR xp9303 (H). Thanks for your help

goebish, 2019-08-23:
Sorry, this git isn't about the Multi Protocol Module, you can ask your question here: https://www.rcgroups.com/forums/showthread.php?2165676-DIY-Multiprotocol-TX-Module

goebish, 2019-08-23:
Also, not every version of the CX10 is compatible with those modules, maybe yours isn't supported (yet).