goebish/nrf24_multipro #52: wltoys12428 v202 protocol

goq1990, 2019-10-11:
can't bind to car, i see on youtube t8sg on v202 bind to this car [https://www.youtube.com/watch?v=6ZRd13hvYt4](url) 
Please help

goebish, 2019-10-11:
The protocol is working (it binds to my V272).
Note this protocol does only work with genuine nrf24l01, ACKnowledgement does not work with clones.

goq1990, 2019-10-11:
I have one more car wltoys a979 bind perfect on this protocol i have nrf24l01+

goebish, 2019-10-12:
I don't understand then ...
I thought it could have been a problem with auto ack but it's already disabled in the code (same as Deviation).

goq1990, 2019-10-13:
Protocol work with car wltoys a979 but with wltoys 12428 crawler no :(

albfan, 2021-06-20:
If you try to bind this type of controller with a T8SG: 

[image] 

follow this: https://github.com/DeviationTX/deviation/issues/1005#issuecomment-864588533

gbraad, 2022-08-12:
Some models have replaced the v2x2 protocol with the XK x450 subprotocol, as also used in the  
12401x/1440xx range. It uses an XN297 and is not fully compatible with an nrf24. Multipro uses the cc2500 for this. https://www.multi-module.org/using-the-module/protocol-details/xk