goebish/nrf24_multipro #57: Eachine E010 Rev 2 Incompatibility Issue

NelsenEW, 2020-12-17:
[image]

Aliexpress link: https://www.aliexpress.com/item/32741187900.html?spm=a2g0s.9042311.0.0.17d44c4defvupA

Hello, I just bought Eachine E010 from Aliexpress. It appears that Eachine has released a newer version of the drone. It has **red** color PCB with _NH010R5_ label as the flight controller. In terms of controller, the throttle and rudder are controlled on the left side of the controller. 

Apparently when I tested out the script for this drone, it does not bind at all. This is a compatibility issue as I have tested the script for the older version with **green** color PCB (_NH010R_ flight controller) and it works perfectly fine. 

I would hope to work this out soon as I have a tight deadline for school project. Would you mind in assisting me to create a new script for this drone or perhaps provide some steps to reverse engineer the protocol such that we can implement a similar method in this version of drone?

Thanks in advance!

goebish, 2020-12-17:
Can you tell which chips are on the quad's pcb and in the remote ?

NelsenEW, 2020-12-17:
[image]
[image]

The part number on the quad is not visible. However, in the remote controller, they seem to be using RF2518 with JR - NH010T5 marking on the pcb. I have searched for RF2518 datasheet on the internet and unfortunately I could not find anything related to RF.

NelsenEW, 2020-12-17:
[image]

Apparently I was able to take a closer look with another drone kit. the ic on the quad is clearly shown which is IC2530.

NelsenEW, 2020-12-17:
Sadly, I do not have the module yet. Is there any other approach?

goebish, 2020-12-17:
another (more complex) approach is with a 2.4GHz capable SDR, eg an HackRF or BladeRF.

NelsenEW, 2020-12-17:
I found out that one of the stores in aliexpress mention that the new version is not backward compatible to the old ones.
The store clearly said that the old ones have been discontinued.

Link: https://www.aliexpress.com/item/1005001639673658.html?spm=a2g0o.detail.1000014.57.76183b76qbEkfc&gps-id=pcDetailBottomMoreOtherSeller&scm=1007.13338.192131.0&scm_id=1007.13338.192131.0&scm-url=1007.13338.192131.0&pvid=48cc5a9e-70a9-4df1-82a8-b80190765fba&_t=gps-id:pcDetailBottomMoreOtherSeller,scm-url:1007.13338.192131.0,pvid:48cc5a9e-70a9-4df1-82a8-b80190765fba,tpp_buckets:668%230%23131923%2387_668%23888%233325%2320_3338%230%23192131%230_3338%233142%239890%232_668%232846%238116%23947_668%232717%237558%23150_668%231000022185%231000066059%230_668%233422%2315392%23527

goebish, 2020-12-17:
@pascallanger have you ever heard of this one ?

pascallanger, 2020-12-17:
I've heard about it but I haven't been able to find someone willing to send me a model for looking at it... I've closed the issue: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/457

By the way, look at this thread on the E016H: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/469
Protocol reversed using SDR only on an unknown RF chip ;-)

pascallanger, 2020-12-17:
If someone here is willing to send his TX+quad or participate for buying one, feel free to contact me.

goebish, 2020-12-17:
@NelsenEW you know what you've to do ;)

NelsenEW, 2020-12-18:
@pascallanger I am willing to send my quad copter, May I know your full address so that I can prepare the shipment?

goebish, 2020-12-18:
github still doesn't allow sending private messages ... go to rcgroups.com and send a private message to hpnuts (pascal)

grostim, 2020-12-25:
@goebish @pascallanger : I have a (Rev02) E010 and an sdr usb key.
Let me know if and how i may help.

pascallanger, 2020-12-25:
Your sdr usb key most likely doesn't go up to 2.4GHz...

grostim, 2020-12-26:
You're probably right. I have RTL-SDR v3. I have never tested it beyond 1090Mhz :)

goebish, 2020-12-26:
RTL-SDR (RTL2832U) goes only up to 1.75 GHz, that's a cheap and nice tool to start experimenting with SDR (ADS-B ;)) but not enough for 2.4 GHz

goebish, 2020-12-26:
Actually you can use a RTL-SDR for 2.4 GHz, by using a down converter:
https://www.rtl-sdr.com/tag/2-4-ghz/
Not sure that's worth it. Also, decoding signals from an unknown chip is not straightforward.

BadRaven, 2021-01-07:
Can I try for some clarity here?  I have over 50 of these Quads in a school club I operate (when there IS school!!), mostly as Racing Hovercraft, so I doubt anyone else has the user experience?

1. Its a little upsetting to read that nobody was prepared to send a quad for investigation as I offered exactly that, but was advised to fund a purchase instead. I was also happy to do that BUT it was not certain at the time that a supply would BE the latest version, so I said I'd wait for two more to arrive I had ordered to be sure that one bought would be the right variant and not a waste  of time and money. Obviously Christmas delayed this but I have now received the two (on differing days) from Banggood and they are both REV 5.

2. Please don't identify the latest board as REV2 alone, it is REV2 of the **REV 5**!  The major revision is the last two characters of the longer code on the boards.

3. Rev 1-3 are green boards, Rev 4 is very rare but is blue, Rev 5's are red boards. (So Far!!).      Rev 1-3 interchange and work to the existing MM code.     Rev 4 (I only have one out of 50 plus!!) seems to be the same format as Rev 5.     Rev 5 will not work with any Rev 1-3.

4. Banggood are selling "spare" receiver boards, pictured as green Rev3, sent as red Rev5, so not useable by anyone who does not have a Rev5 Transmitter.

5. There is a low limit as to the max numbers that can bind and work at once, appears to be 8. Anyone confirm?

goebish, 2021-01-07:
1. just contact hpnuts (Pascal) on rcgroups, I'll give you his address so you send one to him
5. there are only 16 fixed IDs for this protocol but the one used by a module is chosen arbitrarily

BadRaven, 2021-01-07:
Most of the Tx here are only modded to make the right stick left right as Yaw for "car" type steering Mode 2, throttle left, steering right.

It appears from my thread elsewhere that someone has sent now. I have his RCG contact. If he still wants and I now doubt, he can contact me.

With schools shut, I have all the school owned models here, so what ACTUALLY can I do to help.  ;))

pascallanger, 2021-01-07:
@BadRaven
1. you told me that you would recontact me on new year on your latest message on RCGroups so I was waiting. Looking at the price of the quad it looked to be cheaper/easier to just find rather than sending but up to you (consider the shipping back if you need the quad back)
5. A long time ago I've extended the number of IDs using an algorithm not dumps but no one was interested to test it so it never got introduced in the main code...

BadRaven, 2021-01-07:
Well as said above they have now finally arrived, and are Rev 5 so it looks fairly safe (famous last words) to buy one from Banggood China now direct. Other sources might still supply older stock.

1.

So, I'll send money if nobody else has contributed. Advise on Email I passed to you via RCG the latest amount (as it alters by the day and country!) you require via RCG for Eachine E010 from Banggood inc postage, etc and what Email address to pay to.  Looking from UK as I do would not maybe show the same amount, and certainly not the same currency exchange.

5.

The "too many at once to bind" matter is not really an issue even for me, as I try to ensure that only the kids in that race are turned on, stops them flattening batteries messing around in pits and getting models trodden on!!   If its 16 I certainly have bumped into it a few time at what seemed lower numbers on than 16, anyway, just have to get more aggressive with kids to make them turn them off when not racing!  My normal numbers present are 20 plus but only usually max 6 per race. I use a transponder system to lap count, etc.

Actually, I wonder if the marked change with Rev 5 might mean Rev5 has more or at least differing to Rev 1-3. the bind process is certainly different in terms of light indication. It would be useful to know what the number was though.

At  the moment I only have six Rev5's available at once, a couple more when schools resume and boys bring theirs in.

BadRaven (Dave Bran/Bravedan)

pascallanger, 2021-01-07:
Check the progress here: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/457

I don't understand what you mean by this: "Actually, I wonder if the marked change with Rev 5 might mean Rev5 has more or at least differing to Rev 1-3. the bind process is certainly different in terms of light indication. It would be useful to know what the number was though."
Rev5 is completly different from the previous version.

"At the moment I only have six Rev5's available at once, a couple more when schools resume and boys bring theirs in."
The RXs are not important at this stage. What's more needed is the TXs to dump the IDs and frequencies.

taibentre, 2021-09-29:
> [image]
>
> Aliexpress link: https://www.aliexpress.com/item/32741187900.html?spm=a2g0s.9042311.0.0.17d44c4defvupA
>
> Hello, I just bought Eachine E010 from Aliexpress. It appears that Eachine has [...]

My board is Jr-nh010R5DG ? Bind ok with Tx ( red board)

taibentre, 2021-09-29:
[image]
[image]

This board bind ok with tx ( red board) but not ok with irangex irx6 flysky? Help me. Thanks

taibentre, 2021-09-29:
[image]

taibentre, 2021-09-29:
[image]
[image]

BadRaven, 2021-09-29:
As far as I know despite board ident stating REV2, all red boards are Version5. Versions 1-4 (usually green boards but I have seen a Rev4 IN blue!) are OK with green board Tx of any variant. The Red Tx board only works with the red quad board. The light sequence for the newer pairing on bind is different.  An addition was made to the multi-module firmware to add this new variant. Have you got the latest firmware and have you set the "right" E010 module up?

taibentre, 2021-09-29:
I don't know flash irx6. You can share link with me. Thanks

taibentre, 2021-09-29:
[image]

taibentre, 2021-09-29:
[image]
This is irx6

taibentre, 2021-09-29:
Here is my video bind..https://youtu.be/qyGQnK-jdvc

pascallanger, 2021-09-29:
This protocol has been reversed on the multi protocol module: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#e010r5---81
This protocol uses the CYRF6936 RF component which is not available on the nrf24_multipro.

taibentre, 2021-09-29:
Thanks a lot