goebish/nrf24_multipro #58: FliteTest FT Freighter/EZ Power Pack/Banggood C17 support

marktbaldridge, 2020-12-22:
I have a C17 airplane from Banggood (the same is also sold by FliteTest as the FT Freighter, so I added that for searchability) which has a XNS1042CV in the controller. It is the only chip and the antenna is wired in, so it is probably an MCU+RF chip.

I would like to reverse engineer the protocol and get support added to the nrf_multipro software.

Hopefully I have the requisite skills. I am an engineer by trade. :) I have an extra NRF24 module. Can that be put into sniffing mode for this chip? I've done a bit of reading, but haven't seen simple Arduino code yet.  I'm looking for some direction. Is this possible without an SDR?

goebish, 2020-12-22:
Hi,

A nrf24l01 can be used to dump packets sent by a xn297 but the nrf24 multipro project doesn't have the code to do that.
(hint: use a 3 byte address with value 0x55, 0x0F, 0x71, find a channel / bitrate that receives data then decode it since most of the time xn297 whitening is enabled).

The DYI 4-in-1 Multiprotocol module can dump xn297:
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Multiprotocol/XN297Dump_nrf24l01.ino

Or you can use a SDR:
https://github.com/goebish/XN297_dumper

Or a DeviationTX radio:
https://www.youtube.com/watch?v=pWWmSvUSex4

@pascallanger I'm not sure but I think you've already worked on C17 ?

pascallanger, 2020-12-22:
Yes the C17 v1 and v2 protocols are already reversed:
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#GD00X---47
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Multiprotocol/GD00X_nrf24l01.ino