goebish/nrf24_multipro #59: nrf24l01 as receiver

emielsteerneman, 2021-03-10:
If I understand this library correctly, it allows for an nrf24l01 to be placed in a Taranis as an external module, and it supports multiple protocols. I was wondering if it is possible to use a nrf24l01 as a receiver. I have a tx16s controller supporting multiple protocols implemented here, and I would like to communicate with a nrf24l01 at the receiving end.

I haven't found another nrf24l01 library as detailed as this one, I am kind of at a loss. Is it even possible to have an nrf24l01 at the receiving end? Thanks.

goebish, 2021-03-11:
It is possible, take a look at this for example:
https://github.com/execuc/v202-receiver

emielsteerneman, 2021-03-11:
That's great, thanks! Turns out I haven't been looking hard enough after all. Do you know of any other libraries like these for the nrf24l01? Regardless, I am happy. The "issue" can be closed as far as I am concerned.

goebish, 2021-03-11:
The DIY Multiprotocol Module (the one you've in your tx16s) implements a few receivers, BayangRX is using the nrf24l01
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Multiprotocol/Bayang_Rx_nrf24l01.ino

goebish, 2021-03-11:
The DIY Multiprotocol Module (the one you've in your tx16s) implements a few receivers for wireless training, BayangRX is using the nrf24l01
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Multiprotocol/Bayang_Rx_nrf24l01.ino

closed 2021-03-11 by goebish