goebish/nrf24_multipro #63: How to check if firmware loaded is working? Is there a simple way to debug?

randuwiino, 2024-05-18:
Hi Goebish,

I tried to replicate using pro mini 3.3v + NRF24L01+PA+LNA (10uf capacitor 3.3&GND), i've used a 3.3v regulator to power both and it works fine, i have tested using a test transmit receive code and it works fine, now I changed the TAER to AETR in the code and uploaded it, i put a 10K resistor for my ppm, i'm using a jumper t-lite with the external adapter for external module, i soldered the ppm, power and ground, so the module is powered by the transmitter, but now i am trying to bind to a wltoys and it does not bind, everything loos fine, power and also booting, the promini led blinks at power up.

Is there a way i can know if the code is working, any simple debugging i can add to the code? or any sign from the LED?

Thanks very much.

goebish, 2024-05-18:
That's probably because PPM output (from transmitter) shouldn't be directly connected to PPM input (arduino).
You have to shift the level to 3.3V using a transistor or mosfet and a couple of resistors, look for "level shifter circuit". Except if you're sure the PPM output of the transmitter already outputs a 3.3V signal, but I doubt it.

goebish, 2024-05-18:
[image]

randuwiino, 2024-05-18:
oh i see, that make sense, i have a logic level converter, i will try, are there any signal in LED that i need to watch for to see if its working or not? also i noticed the board does not turned on when the throttle stick is not at the down position, so in my case which is the WLTOYS, the instructions says only elevator up, so it means throttle down + elevator up? is that correct?

goebish, 2024-05-18:
A logic level converter might work but is not ideal because you'll need a voltage source of the same voltage as the PPM output (to be measured with a scope ...). Better use a mosfet or transistor so only a 3.3V source is required, which you already have.
If the instructions says only elevator up, the other stick needs to be centered on both axis, not down, this selects the protocol, then the LED will stop blinking but the protocol will only start when throttle is down (that's for safety).

goebish, 2024-05-18:
And if you want to debug, start by checking if you're stuck here, that's probably the case:
https://github.com/goebish/nrf24_multipro/blob/45fe93f8a575273dbe0e602066a8964e33543a70/nRF24_multipro/nRF24_multipro.ino#L243
Just add some debug output code into the { }, as long as the code is here it means PPM input is not detected.

goebish, 2024-05-18:
don't use a simple Serial.println, that would mess up the PPM interrupt, use something like that into the { }:

```
static unsigned long debugStamp = millis();
if (millis() >= debugStamp) {
    Serial.println("Waiting for PPM input...");
    debugStamp = millis() + 1000;
}
```

randuwiino, 2024-05-18:
I don't have a mosfet, my hands are too shaky to unsolder too small components, but i do have an NPN transistor, can you show me a diagram if i use NPN transistor? Thanks very much for the debug code, that would help a lot.

goebish, 2024-05-18:
Basically it's the same as with a MOSFET, with a NPN transistor you would:
- connect its Collector to ground (bottom pin in the diagram)
- connect its Base to the PPM output of the transmitter with a 10k resistor in serial (left side pin in the diagram)
- connect its Emitter to the arduino PPM interrupt pin and to +3.3V with a 10k resistor in serial (top pin in the diagram)
Use google to know which pins C, B and E are with your transistor's model.

goebish, 2024-05-18:
Sorry, I edited my previous message, you might have connected your transistor in reverse...

randuwiino, 2024-05-18:
i added the transistor and resistors as per the diagram, now when i power up the transmitter, led blinks once. is that the correct behavior? although i still could not bind.

goebish, 2024-05-18:
I just had a look at the (old) code and the LED is not used actually, first try to add some debug code to check if PPM input is detected.

randuwiino, 2024-05-18:
[image]
[image]
I added a debug code, i get the Serial.println from the setup, but not the LED HIGH inside the Loop.