goebish/nrf24_multipro #65: no connection to E010

druckgott, 2025-02-08:
Hi I have a Jumper T15 and connect alle the things this the arduino nano and the nRF24 modul.

I select Multimodul in the jumper on external modul and did the things like here:
https://www.instructables.com/NRF24-Jr-Module-for-Toy-Quadcopters/
Its connectet like in the picture:

[image]

I only have 4 channels aktive.
On the startup I do this:
Rudder right + Aileron right + Elevator down = EAchine E010, NiHui NH-010, JJRC H36 mini

I only get one light on the Arudino nano LED and it´s on.

druckgott, 2025-02-08:
I think thats my problem:
https://github.com/goebish/nrf24_multipro/issues/57
I have the red pcb version

goebish, 2025-02-09:
The code currently works only for the first revision of the E010, and even then, it doesn’t always function reliably. This is because emulating an XN297 with an NRF24L01 at 250kbps is not perfect. For better performance and compatibility using a CC2500 instead is necessary (with different xn297 emulation layer).

Additionally, please note that nrf24_multipro is an obsolete project and is no longer maintained. For a more up-to-date and actively maintained solution, I suggest checking out the following repository:

https://github.com/pascallanger/DIY-Multiprotocol-TX-Module

druckgott, 2025-02-09:
Ok thx

closed 2025-02-09 by druckgott